openwisp/openwisp-controller #809: [bug] Custom command expects string type

mips171, 2023-08-26:
The controller expects strings, but if you pass it a list, we get a TypeError.
For instance, if you define a custom command that has checkboxes like:

```
(
        'lockbands',
        {
            'label': 'Lock to Specific Bands',
            'schema': {
                'title': 'Lock Mobile Bands',
                'type': 'object',
                'properties': {
                    'bands': {
                        'type': 'array',
                        'title': 'Warning: Check that the bands you are locking to are actually available first or your device might not reconnect - Select Bands to Lock',
                        'items': {
                            'type': 'string',
                            'enum': [
                                '4G Band 1',
                                '4G Band 3',
                                '4G Band 5',
                                '4G Band 7',
                                '4G Band 8',
                                '4G Band 28',
                                '4G Band 40',
                            ]
                        },
                        'uniqueItems': True,
                        'minItems': 1
                    },
                    'modem_index': {
                        'type': 'string',
                        'title': 'Modem ID (leave blank for default)',
                    },
                },
                'required': ['bands'],
                'additionalProperties': False,
            },
            'callable': lockbands_command_callable,
        }
    ),
```

TypeError: sequence item 0: expected str instance, list found

There are a ton of ways to fix this, and I'm not sure if there are other architecture-level decisions that would limit the choices, but what I did is cast the argument to a string
On line 606 in `openwisp_controller/connection/base/models.py`
replace
    `return ', '.join(self.arguments)`
with
` return ', '.join(str(arg) for arg in self.arguments)`

nemesifier, 2023-08-29:
So if my understanding is correct, right now we only allow simple string input values and allowing also lists would improve the solution?

itsAyush5, 2025-04-03:
Hi, I would like to work on this issue. Can you assign it to me?

nemesifier, 2025-04-03:
@itsAyush5 there's no need to assign issues here, just coordinate with us in the [dev chat](https://matrix.to/#/#openwisp_development:gitter.im) and open a PR when you're ready, thanks! 🙏