openwisp/openwisp-controller #813: undocumented endpoints and swagger API access

dberardo-com, 2023-10-22:
i was going through the openwisp-config script and found out that the core endpoints, such as download-configuration: https://github.com/openwisp/openwisp-config/blob/master/openwisp-config/files/openwisp.agent#L920 are undocumented on the openwisp-controller doc. indeed it seems the controller doc is actually using a different API spec: https://github.com/openwisp/openwisp-controller/tree/1.0#download-device-configuration since the config endpoints are flagged as legacy in the code: https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/config/utils.py#L153

why is there a discrepancy between the 2 APIs and how could one access the API documentation currently in use by the controller to serve openwisp-config clients?

also: is there an online version of the swagger API available via the demo: https://openwisp.org/demo.html ?

dberardo-com, 2023-10-22:
also, it seems that some endpoints, such as "checksum" are undocumented in this repo. where to find those ?

nemesifier, 2023-10-23:
The logic is here:
https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/config/utils.py#L113-L180
https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller/config/controller

It's been developed before django-rest-framework and swagger were integrated, it's not REST API but some HTTP endpoints designed specifically for the OpenWrt agent, which retain some logic which was previosuly built in ruby on rails (2008-2016).

dberardo-com, 2023-10-23:
ok, are you planning to release a new version of the agent to adapt to the new API or is it not planned yet?

> t's been developed before django-rest-framework and swagger were integrated, it's not REST API but some HTTP endpoints designed specifically for the OpenWrt agent, which retain some logic which was previosuly built in ruby on rails (2008-2016).

i understand that this API is therefore undocumented right ? is there any hint of what needs to be bundled in the configuration compressed folder ?

closed 2024-05-02 by nemesifier