openwisp/openwisp-controller #828: [feature] Add endpoint that returns co-ordinates on indoor map (floorplan)

pandafy, 2024-02-06:
Add an API endpoint that takes the location ID and returns co-ordinates of all the devices.

- It should also return all available floors in the location. 
- It should be possible to filter the devices based on the floor.

dee077, 2025-01-25:
```
{
 'coordinates': ['-140.38620,40.369227', '-140.38620,40.369227'], // indoor from below
 'floors': [1, 2]
}
```
Is this response okay?

this is the sample response for DeviceLocation model if I need to add more data what other entires should I add from below?
```
{
  "deviceLocation model entries": [
    // Entry 1
    {
      "location": {
        "id": "761b17b6-44d6-4610-b4e7-be0a72554d07",
        "type": "Feature",
        "geometry": {
          "type": "Point",
          "coordinates": [12.512124, 41.898903]
        },
        "properties": {
          "type": "indoor",
          "is_mobile": false,
          "name": "test-location",
          "address": "Via del Corso, Roma, Italia"
        }
      },
      "floorplan": {
        "id": "0656af6d-1828-4ec6-9dc5-4d9a4fc18402",
        "name": "test-location 1st Floor",
        "floor": 1,
        "image": "http://testserver/media/floorplans/0656af6d-1828-4ec6-9dc5-4d9a4fc18402.jpg",
        "created": "2025-01-25T13:00:07.068880+01:00",
        "modified": "2025-01-25T13:00:07.068880+01:00"
      },
      "indoor": "-140.38620,40.369227"
    },
    // Entry 2
    {
      "location": {
        "id": "761b17b6-44d6-4610-b4e7-be0a72554d07",
        "type": "Feature",
        "geometry": {
          "type": "Point",
          "coordinates": [12.512124, 41.898903]
        },
        "properties": {
          "type": "indoor",
          "is_mobile": false,
          "name": "test-location",
          "address": "Via del Corso, Roma, Italia"
        }
      },
      "floorplan": {
        "id": "cd4cf4b6-9314-4613-a2b3-27b464da6398",
        "name": "test-location 2nd Floor",
        "floor": 2,
        "image": "http://testserver/media/floorplans/cd4cf4b6-9314-4613-a2b3-27b464da6398.jpg",
        "created": "2025-01-25T13:00:07.077346+01:00",
        "modified": "2025-01-25T13:00:07.077346+01:00"
      },
      "indoor": "-140.38620,40.369227"
    }
  ],
}
```

dee077, 2025-01-31:
Is this response okay? 
Should I include any other field or data in this, please suggest if so.  
```
{
  "devices": [
    {
      "device_id": "device1-id",
      "floorplan": {
        "id": "01de93a7-8581-4a89-b9c9-bb4afba8935c",
        "floor": 1
      },
      "indoor": "-140.38620,40.369227"
    },
    {
      "device_id": "device2-id",
      "floorplan": {
        "id": "ec5c7ef5-991a-4657-86ca-95b15e3e4ba3",
        "floor": 2
      },
      "indoor": "-140.38620,40.369227"
    }
  ],
  "available_floors": [1, 2]
}
```

dee077, 2025-05-14:
## Appraoch
- The current implementation of the draft PR returns data in JSON format; this should be updated to return GeoJSON format to be compatible with netjsongraph.js for rendering the floorplan map.
- Filter class is already been made.

PR #976 (2025-07-29): [feature] Add API endpoint for indoor map coordinates #828
Implemented API to return device coordinates for a given location ID.

Fixes #828 

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [ ] I have updated the documentation.

## Reference to Existing Issue

Closes #828 

## Description of Changes
- Added endpoint `floorplan_coordinate [... cut]
commit aecda03c (2025-10-07): [feature] Added API endpoint for indoor map coordinates #828

Implemented API to return indoor coordinates of devices for a given location.

Closes #828

---------