openwisp/openwisp-controller #831: [bug] Deleting VPN template does not delete VpnClient objects

pandafy, 2024-02-09:
**How to replicate**

1. Open Django shell (in command line) and check the number of VpnClient objects using this query  `VpnClient.objects.count()`
2. Create a VPN server object with OpenVPN backend 
3. Create a VPN template for this VPN server 
4. Add the VPN template to a device 
5. Go to the VPN template page and delete the Template 
6. Open Django shell (in command line) and check the number of VpnClient objects using this query `VpnClient.objects.count()`

**Expected outcome**
The number of VpnClients is different in **Step 1** and **Step 6** should be same.

**Actual outcome**

The number of VpnClients is different in **Step 1** and **Step 6** (increased by one). It is because the deletion of VPN template didn't trigger deletion of related VpnClient objects (there is no direct relation).

pandafy, 2024-02-09:
I found that it is possible to have two different VPN templates for the same VPN server object. But, it is not possible to apply both the templates on the same device, as the JSONSchema generates the following error

> There is a conflict with the specified templates. Invalid configuration triggered by "#/files", validator says:

[image]

I think we should discourage using different templates for the same VPN server object on the same device using validation (we can create a separate issue for this).

pandafy, 2024-02-09:
We should double check the working of https://github.com/openwisp/openwisp-controller/blob/226442c47be5031e64033ae44c8b00aaf2623c68/openwisp_controller/config/base/config.py#L229-L275

nemesifier, 2025-02-27:
> We should double check the working of the following code. I think, it is not getting triggered when the template object is deleted.
> 
> [openwisp-controller/openwisp_controller/config/base/config.py](https://github.com/openwisp/openwisp-controller/blob/226442c47be5031e64033ae44c8b00aaf2623c68/openwisp_controller/config/base/config.py#L229-L275)

I think you're right!

Let's add a foreign key from VpnClient to Template, we'll need migrations that work like this:

1. add the column and allow it to be NULL
2. fill the column for all vpn clients records
3. disallow the column to be NULL

commit 0bbbd776 (2025-03-18): [fix] Deleting VPN template will delete related objects #831

Deleting VPN template will delete related VpnClient objects.

Fixes #831
commit 5186ad99 (2025-03-20): [change/fix] Added relation from VpnClient objects to Template #831

Make sure VpnClient objects are deleted when VPN-client templates are deleted.

Fixes #831
closed 2025-03-20 by nemesifier