openwisp/openwisp-controller #835: SEND COMMAND option not displayed when it should be

csyost85, 2024-02-11:
https://github.com/openwisp/openwisp-controller/blob/ca86ceda3280fca15b535fa25cbeb7a3d90b62c4/openwisp_controller/connection/static/connection/js/commands.js#L9

the SEND COMMAND option is not displayed even though it should be.
Caused by incorrect logic in command.js

Logic starting at line 9

    if((typeof(owControllerApiHost) !== 'undefined')|| isDeviceRecoverForm()) {
        return;
    }
 
typeof(owControllerApiHost) !== 'undefined' should be true for it to work however line 9 will cause the display code to exit ff true.  Recommend updating logic, or change back to previous logic
if (isDeviceRecoverForm())

nemesifier, 2024-02-12:
Hi @csyost85, we haven't noticed issues with this logic, could you give us steps to reproduce the issue? Or is this reproducible on the [openwisp demo system](https://openwisp.org/demo.html)?

nemesifier, 2024-05-02:
This should be fixed in the current master.

closed 2024-05-02 by nemesifier