openwisp/openwisp-controller #848: [change/admin] Format configuration and configuration variables as indented JSON if the user has only view permission

pandafy, 2024-04-01:
The JSONSchema editor is not shown when the user only has view permission. This also hides the configuration of the device. Moreover, the configuration variables are shown as string representation of Python object: 

[image]

This is not good for UX. We shall show the value of these fields as indented JSON in the UI when the user has read-only permission. 

This issue affect both template and device admin pages.

------------------------------------------------------------

This issue was first found in https://github.com/openwisp/openwisp-controller/pull/840#pullrequestreview-1923292713

nemesifier, 2024-11-13:
Would be good to address this as well as https://github.com/openwisp/openwisp-controller/issues/560 to improve read-only capabilities.

Utkarsh09102004, 2025-03-14:
Could you please provide detailed steps to reproduce this issue?

nemesifier, 2025-05-27:
Addition: this shall be done for Device, Template and Vpn objects, we shall fix this for both `config` and `context`.

@Utkarsh09102004 the issue is visible when accessing the admin with an user who has only read permissions.

pandafy, 2025-07-14:
Django handles rendering of read-only fields differently. Instead of using the widget for the field, it [handles the rendering based on field type](https://github.com/django/django/blob/ac2d907f459b2cb7ddea7451764e7020ffee08e6/django/contrib/admin/utils.py#L431-L469). 

In https://github.com/openwisp/django-loci/pull/158/, we relied on the undocumented workaround of setting the `read_only` attribute on the form widget to `True` which tricks Django to use the widget for rendering the field. 

This workaround has been removed in https://github.com/django/django/pull/19063 which is merged in Django main. So, relying on this method would not work in Django 6.X unless a change is made in Django. 

Another option is to extend the `django.contrib.admin.helpers.AdminReadonlyField` and then monkey patch `django.contrib.admin.helpers.Fieldline` to use the extended field. 

There was a [ticket](https://code.djangoproject.com/ticket/30577) opened in Django 6 years ago for adding this functionality. The ticket was closed and the discussion was moved to the [Django forum](https://forum.djangoproject.com/t/feature-request-discussion-custom-rendering-for-readonly-fields-in-admin/32009/2), but it hasn't gained any traction.

Kapingacoder, 2026-03-30:
Hi everyone,

I have been studying this issue and the related PRs to better understand the UX problem with read-only JSON/dict fields in the Django admin. From what I gathered, the main difficulty comes from Django’s handling of readonly fields and the removal of the previous widget-based workaround in newer Django versions.

It seems the current direction is moving toward a more robust solution that does not rely on undocumented behavior, which makes sense for long-term maintainability.

I am currently trying to deepen my understanding of the OpenWISP controller codebase and Django admin customization, so this discussion is very helpful.

I would like to ask:
Is the approach in PR #1327 considered the final direction for this issue, or are there still improvements expected (for example better formatting, validation visibility, or readability improvements for complex JSON)?

If useful, I would also be happy to help with testing, reviewing UX consistency across admin sections, or contributing small improvements.

Thank you for the great work and discussion.

commit 728ab658 (2025-07-31): [fix:ux] Fixed rendering of JSON fields in readonly mode #848

Fixes #848
commit c92793d9 (2025-08-07): [fix:ux] Fixed rendering of JSON fields in readonly mode #848

Fixes #848
commit 0f2d7a4a (2025-08-07): Squashed commit of the following:

commit c28442a7ab3834cf0cbf5347c7935fc45f1dab65
Author: Gagan Deep <pandafy.dev@gmail.com>
Date:   Thu Aug 7 17:36:01 2025 +0530

    [tests] Fixed failing tests

commit c92793d97cb227dc3398f97f572fa015a4a7abee
Author: Gagan Deep <pandafy.dev@gmail.com>
Date:   Tue Jul 29 22:55:22 2025 +0530

    [fix:ux] Fixed rendering of JSON fields in readonly mode #848

    Fixes #848

commit 3817db7815a68c240defe5679048bf4341e37237
Author: Gagan Deep <pandafy.dev@gmail.com>
Date:   Sat Aug 2 03:35:22 2025 +0530

    [docs] Added changelog for Version 1.1.2

commit 6d997 [... cut]