openwisp/openwisp-controller #864: [bug] Multiple SubnetDivisionRule of different sizes

pandafy, 2024-05-30:
**Steps to replicate**

1. Ensure you have at least one device for the `default` organization in OpenWISP
2. Create a subnet `10.0.0.0/16` 
3. Add a SubnetDivisionRule as follows: 
    Label: DEV1
    Organization: Default
    Size of subnet: 31
    Number of subnet: 1
    Number of IPs: 1
4. Ensure that subnets and IPs are created for the exsiting device
5. Add another SubnetDivisionRule as follows: 
    Label: DEV2
    Organization: Default
    Size of subnet: 29
    Number of subnet: 1
    Number of IPs : 8
    
**Expected Outcome**

IPs and Subnets are provisioned for the device according to the DEV2 subnet divison rule. 

**Actual Outcome**

Only IPs and Subnets are created for DEV1

pandafy, 2024-05-30:
I suspect this occurs because of the `Reserved Subnet`. The size of the reserve subnet is dictating the size of the subsequent subnets. 

If we change the order of created of subnet division rules, i.e. create DEV2 before DEV1, then the subnet created for DEV1 would be /29 instead of /31

sicaario, 2025-12-29:
Hi @pandafy and @nemesifier!

I'd like to work on this issue. Based on the description, it seems the problem is with the Reserved Subnet size affecting subsequent subnet divisions.

## My Approach:

1. **Understand the Current Logic**: First, I'll examine the subnet division rule implementation in the codebase to understand how reserved subnets are currently handled and how they interact with multiple division rules.

2. **Reproduce the Issue**: I'll follow the exact steps mentioned to reproduce the bug locally and verify the behavior.

3. **Investigate the Root Cause**: Based on the observation that changing the order of rule creation affects the outcome, I'll investigate:
   - How the reserved subnet size is calculated
   - How subsequent division rules use or are affected by the reserved subnet
   - Why the first rule's subnet size is dictating subsequent rules

4. **Implement a Fix**: The fix should ensure that:
   - Each SubnetDivisionRule creates subnets of its specified size independently
   - Reserved subnets don't interfere with the proper allocation of different-sized subnets
   - The order of rule creation doesn't affect the outcome

5. **Write Tests**: Add comprehensive test cases to cover:
   - Multiple division rules with different subnet sizes
   - Different orders of rule creation
   - Edge cases with various subnet size combinations

6. **Documentation**: Update any relevant documentation if needed.

Could one of the mentors please assign this issue to me? I'm excited to contribute and learn from this project!

Thank you!