openwisp/openwisp-controller #894: [feature] Keep track of changes with django-reversion

nemesifier, 2024-07-24:
**Is your feature request related to a problem? Please describe.**

It seems to me that changing objects like devices or templates, which are monitored in the admin via [django-reversion
](https://github.com/etianen/django-reversion), doesn't store a revision.

Therefore if a device is changed via the API, there's no way to rollback changes.

**Describe the solution you'd like**

Ideally, for each model which supports reversion in the admin, we would need to add equivalent features for the REST API:

- A way to save a revision whenever a change is made via PUT/PATCH
- An endpoint that lists the revisions
- An endpoint which allows to inspect a revision
- A way to restore a revision

**Describe alternatives you've considered**
There may be third party open source tools we can use, otherwise we can add our own logic in openwisp-utils and start rolling this feature out in the main modules.

okraits, 2024-08-12:
A revision should possibly be created if an object is saved by anything but the admin.

https://django-reversion.readthedocs.io/en/latest/api.html#creating-revisions

Additionally it might be need to use `follow()` when registering a model:

https://django-reversion.readthedocs.io/en/latest/api.html#registration-api

nemesifier, 2024-08-12:
@okraits reading the issue again, I think I forgot to specify that I meant to convey that this problems happen when modifying objects via the REST API.

dee077, 2025-01-13:
Hi @nemesifier @pandafy 
To implement revision tracking for API endpoints and models, there are several ways to implement the logic. Below are four approaches I came up with their respective pros and cons.

---

### 1. **Centralized Logic for all the View Classes**

- Create a class with logic for the reversion and use it as base class for all the view classes (will override these methods `perform_create`, `perform_update`, `perform_destroy`).
- Allows precise control over revision tracking for each view and provide reversion with proper context like what endpoint or which user cause the change.
- Revision logic may need to be repeated across multiple views.
- Can become harder to maintain as the number of views increases.

---

### 2. **Using Middleware**

- Ensures that the current user context is available across all views and models.
- Keeps revision tracking logic separate from views and models.
- Adds additional processing for each request, which can affect performance.
- Does not directly access the objects being tracked, so revision tracking is more generalized.

---

### 3. **Using Signals**

- Handles revision tracking independently from views and models, keeping things clean.
- Revision tracking is triggered automatically on create, update, or delete actions.
- Signals can make debugging harder and introduce hidden behavior.
- Incorrect signal handling can cause side effects in unrelated parts of the system.

---

### 4. **Centralized Logic in Models**

- Create a base class for all the models required reversion which will track all changes through overriding the `create` and `update` methods.
- Needed extra logic for the context of the change like user, endpoint, request type etc.
- Models become tightly coupled with the base class, reducing flexibility.

---

#### Which Approach to Choose?
or there is a better solution for this scenario? 

Please let me know your thoughts.

pandafy, 2025-01-24:
Thanks for proposing multiple solutions @dee077. I checked the documentation of django-reversion and found [`reversion.views.RevisionMixin`](https://django-reversion.readthedocs.io/en/latest/views.html#reversion-views-revisionmixin). Maybe, we can use this on the API views.

shwetd19, 2025-02-23:
Hi @dee077, @okraits, and @pandafy,  

We can integrate `reversion.views.RevisionMixin` into our API views to automatically wrap every request in a revision block... this approach is designed for class-based views and handles revision creation properly... ,including associating the request user with the revision metadata. 

### Notes : 
1. By inheriting from `RevisionMixin`, every non-safe request (e.g., `PUT`, `PATCH`, `POST`) will automatically be wrapped in a revision block, ensuring changes are tracked without extra boilerplate.  
2. The request user is automatically added to the revision metadata, solving the "who changed it" problem.  
3. We can use `reversion.set_comment()` in specific methods (e.g., `perform_create`, `perform_update`) to add meaningful revision comments.  
   - The `revision_manage_manually` attribute can be set to `False` (default) to ensure revisions are always saved, or toggled if we need manual control in specific cases.  
   - The `revision_using` attribute allows us to specify a database for revision data if we need to separate it from the default.  
   - We can override `revision_request_creates_revision(request)` to fine-tune which requests trigger revisions (e.g., exclude certain methods if needed).  
4. This approach keeps the logic within the view layer, avoiding the need for middleware or model-level changes, and uses.. `django-reversion`’s built-in functionality.  

What do you think about this approach? Does it align with the project’s needs, or is there something else we should consider? Looking forward to your feedback!

dee077, 2025-02-24:
I am working on this will make a draft PR soon.