openwisp/openwisp-controller #902: [bug] Flaky Selenium Test: test_unsaved_changes

nemesifier, 2024-08-01:
Flaky test:

```
======================================================================
FAIL: test_unsaved_changes (openwisp_controller.config.tests.test_selenium.TestDeviceAdmin) [Alert should be displayed after making changes]
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openwisp-controller/openwisp-controller/openwisp_controller/config/tests/test_selenium.py", line 152, in test_unsaved_changes
    WebDriverWait(self.web_driver, 1).until(EC.alert_is_present())
  File "/opt/hostedtoolcache/Python/3.8.18/x64/lib/python3.8/site-packages/selenium/webdriver/support/wait.py", line 105, in until
    raise TimeoutException(message, screen, stacktrace)
selenium.common.exceptions.TimeoutException: Message: 

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.8.18/x64/lib/python3.8/unittest/case.py", line 60, in testPartExecutor
    yield
  File "/opt/hostedtoolcache/Python/3.8.18/x64/lib/python3.8/unittest/case.py", line 582, in subTest
    yield
  File "/home/runner/work/openwisp-controller/openwisp-controller/openwisp_controller/config/tests/test_selenium.py", line 154, in test_unsaved_changes
    self.fail('Timed out wating for unsaved changes alert')
  File "/opt/hostedtoolcache/Python/3.8.18/x64/lib/python3.8/unittest/case.py", line 753, in fail
    raise self.failureException(msg)
AssertionError: Timed out wating for unsaved changes alert
```

nemesifier, 2024-08-02:
This problem is caused by this change in Chromium:
https://chromium.googlesource.com/chromium/src/+/5b3168e8f8c4487320c654dd761622b89d548514.

nemesifier, 2024-08-02:
It seems that we may have to switch using bidirectional sessions with the web driver, but it's not clear to me right now how to do that. A temporary workaround could be to use a chromium web-driver version older than 126.

dcki, 2025-04-08:
Hi, I was trying to fix this in something totally unrelated, and encountered this PR on the way to an answer. Here is the answer I found. I hope it helps (and I think it probably will):

https://issues.chromium.org/issues/351858989#comment30

commit e5442429 (2024-08-02): [tests] Fixed flaky selenium test: test_unsaved_changes #902

Fixes #902