openwisp/openwisp-controller #904: [bug] Selenium tests contain JS errors due to failed websocket connection

nemesifier, 2024-08-02:
Selenium tests contain JS errors due to failed websocket connection.

We need to switch from django's `StaticLiveServerTestCase` to `ChannelsLiveServerTestCase`.
I also recommend making the base selenium class in openwisp-utils automatically inherit `ChannelsLiveServerTestCase` and enforce this across OpenWISP.

We also have to figure out:

- How to automatically assign the debugging port (it seems ChannelsLiveServerTestCase doesn't define the port during class initialization)
- How to use BiDi sessions everywhere (due to Classic sessions being limited, see https://chromium.googlesource.com/chromium/src/+/5b3168e8f8c4487320c654dd761622b89d548514)

dee077, 2025-04-15:
I have updated the selenium test in utils to inherit from `ChannelsLiveServerTestCase` instead of `StaticLiveServerTestCase` and found it is failing on Django 5.2 and passing on others.

This open issue in [channels repo](https://github.com/django/channels/issues/2148) as 5.2 is added recently.

1. Should all the selenium tests across all modules inherit from `ChannelsLiveServerTestCase`? 
2. We can't run the test case in parallel after this, we have to separate the unit test and selenium test.
#### And how should we proceed then?

pandafy, 2025-04-28:
> * Should all the selenium tests across all modules inherit from `ChannelsLiveServerTestCase`?

Yes

> * We can't run the test case in parallel after this, we have to separate the unit test and selenium test.

Can you please explain why?

dee077, 2025-04-28:
> Can you please explain why?

When running tests using `StaticLiveServerTestCase` in parallel, Django creates a new process for each test worker. Inside each of these processes, a thread is started to run a `WSGI server` for serving the application, along with additional threads for executing the individual tests.

However, when running tests with `ChannelsLiveServerTestCase` in parallel, a similar initial process is created for each test worker. But instead of spawning a thread for the server, `ChannelsLiveServerTestCase` starts a new **Daphne (ASGI) server as a separate process not as a thread**. Along with this, threads are also spawned to execute the individual tests.

This causes a problem:
Python does not allow a daemonic process (the worker running for tests) to create child processes (the Daphne server process). As a result, an error occurs:
```
======================================================================
ERROR: test_menu_on_wide_screen (test_project.tests.test_selenium.TestMenu.test_menu_on_wide_screen)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/deepanshu/Documents/GSOC/OpenWisp/.utils/lib/python3.12/site-packages/django/test/testcases.py", line 284, in _setup_and_call
    self._pre_setup()
  File "/home/deepanshu/Documents/GSOC/OpenWisp/.utils/lib/python3.12/site-packages/channels/testing/live.py", line 61, in _pre_setup
    self._server_process.start()
    ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/multiprocessing/process.py", line 118, in start
    assert not _current_process._config.get('daemon'), \
    ^^^^^^^^^^^^^^^^^
AssertionError: daemonic processes are not allowed to have children
```
### Solutions:
1. Don't use --parallel
2. Start the Dapine server manually in a different terminal. (work locally but can cause issues in ci).

#### Let me know if there are any other solutions!

pandafy, 2025-04-28:
@dee077 can you open a draft PR in openwisp-utils? I would like to do some debugging before finalizing on a solution.

AnirudhSinghRajora, 2026-10-29:
Hey, This issue has been stale for a quite so I am working on this :)