openwisp/openwisp-controller #910: [bug] Multiple UI issues in configuration editor for Device

kkreitmair, 2024-08-21:
**Describe the bug**
There are two bugs in the configuration default editor for a Device in the admin.
1.  For select fields, with predefined values in the corresponding JSON schema, when a value is saved other than the default, the field will be empty after saving. The saved value is though present in the advanced mode editor. This happens only, in combination of the lib `select2`. See also corresponding GIF below.
2. The dropdown for `Configuration Menu`  and `Object Properties` are not shown anymore, if a configuration section is saved, which has in its JSON schema definition a property with type array and should have unique items (such as [Mwan3](https://github.com/openwisp/netjsonconfig/pull/232)) . See also corresponding GIF below.

**Steps To Reproduce**
Steps to reproduce the behavior for the **1.** Issue:
1. Go to the change view of an existing Device.
2. Select the Configuration tab.
3. Create a new Configuration for the section `General`.
4. In the field timezone, set a value other than the default.
5. Click `Save and continue editing`.
6. Scroll down to the configuration section `General`. The previous selected and saved value is not shown.

Steps to reproduce the behavior for the **2.** Issue:
1. Go to the change view of an existing Device.
3. Select the Configuration tab.
4. Create a new Configuration for the section `Mwan3`. (I patched this into netjsonconfig from https://github.com/openwisp/netjsonconfig/pull/232 because it has a property in its schema definition with type array and allows only unique items)
5. Click `Save and continue editing`.
6. Scroll down to the configuration section and click `Configuration Menu`. The menu dropdown will not show up, but all configuration fields will be disabled.

**Expected behavior**
Expected behavior for the **1.** Issue:
The saved value, should always be displayed.

Expected behavior for the **2.** Issue:
The dropdown menu of the `Configuration Menu` and the `Object Properties` should be always displayed.

**Screenshots**
GIF for showing the **1.** Issue: 
[image]

GIF for showing the **2.** Issue:  
[image]

**System Information:**
 - OS: Debian 12
 - Python Version: 3.9.2
 - Django Version: 4.2.15
 - Browser and Browser Version (if applicable): Firefox and Chromium

kkreitmair, 2024-08-21:
I will open a PR for the bug fix in the next days.

nemesifier, 2024-08-21:
Thank you very much for reporting this @okraits @kkreitmair, I didn't notice this yet.

PR #913 (2024-09-05): [fix] Config editor issues
## Checklist

- [X] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [X] I have manually tested the changes proposed in this pull request.
- [ ] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [ ] I have updated the documentation.

## Reference to Existing Issue

Closes #910.

## Description of Changes

To address the first Issue described in #910, an update of the way the selects are initialized by the jsonschema-ui.js is needed.

To address the second  [... cut]
commit 23025093 (2024-09-05): [fix] Fix object and config menu not opening in Device config editor #910

This addresses the problem, that the drop-down for Configuration Menu
and Object Properties are not shown anymore in the device config
editor, if a configuration section is saved, which has in its JSON
schema definition a property with type array and should have unique
items. The cause of the problem is a call to super, when there is no
super. This commit will fix this issue by removing the call to super,
because it is not needed in this special case neither in all other
cases.

Fixes #910
commit eb1d72fb (2024-09-05): [fix] Fix select fields in Device config editor #910

For select fields, with predefined values in the corresponding JSON
schema, when a value is saved other than the default, the field will
be empty after saving in the Device config editor. This happens only,
in combination of the lib select2.  The cause of the issue is, that
the jsonschema-ui.js version does use a deprecated way to initialize
select2 inputs. This commit will fix this issue by updating the way of
initializing the inputs.

Fixes #910
closed 2024-09-05 by nemesifier