openwisp/openwisp-controller #949: [bug] Unable to delete offline device

pandafy, 2024-12-06:
**Describe the bug**

Devices that are no longer connected to OpenWISP cannot be deleted.

This issue occurs because, in #840, we introduced a check requiring a device's config status to be set to `deactivated` before allowing deletion. However, this prevents users from deleting disconnected devices that cannot update their config status.

**Steps To Reproduce**

1. Navigate to the Devices section.
2. Select a device that is no longer connected to OpenWISP.
3. Attempt to delete the device.

**Expected behavior**
It should be possible to delete the device

pandafy, 2024-12-06:
**Possible solution**

1. Make the Delete Button Visible: Allow the delete button to remain visible for devices with a deactivating config status.
2. Intermediate Warning Page: When the user clicks the delete button, display an intermediate page with the following warning:

> "The device is still in the deactivating state, meaning its configuration is still present on the device. If you wish to remove the configuration from the device, please wait until the config status changes to deactivated. Proceeding will delete the device from OpenWISP without ensuring its configuration has been removed."

akhilsharmaa, 2024-12-09:
I am working on this issue.

akhilsharmaa, 2024-12-12:
>  deleting disconnected devices that cannot update their config status.

hey @pandafy, i have a doubt "how exactly we will know that device is disconnected ?" or " how do we know that this specific device cannot update it's `config status`"  ?  

Do we need to send a basic command to the device, if it fail to execute that command then we consider that device as disconnected ?

pandafy, 2024-12-13:
> Do we need to send a basic command to the device, if it fail to execute that command then we consider that device as disconnected ?

No, we only need to provide a way to override the current flow for device deletion. 

**Current flow:**

1. Click "Deactivate" on the device page.
2. Wait for the device to apply the configuration removal and report back.
3. Once confirmed, the user can delete the device.

**Problem:**
If the device is disconnected during step 2, OpenWISP cannot confirm that the configuration is removed, so the device cannot be deleted.

akhilsharmaa, 2024-12-16:
i got it @pandafy, 
unfortunetly, I couldn't reproduce the issue, everytime i deactivate the device and then stop  the router instantly (vbox), either it deactivate successfully or give the popup: 
[image]
and after this popup device deactivating successfully, 
maybe because i'm using the vm, so.. 

**Anyone want to work on this please go ahead. i am not working on this.**

pandafy, 2024-12-16:
@akhilsharmaa stop the VM before deactivating the device.

akhilsharmaa, 2024-12-16:
> [@akhilsharmaa](https://github.com/akhilsharmaa) stop the VM before deactivating the device.

@pandafy,  i tried deactivating the vm after stopping the vm. but still i can delete that device. please this screen-rec : 

in the video, you can see that vm is connected to the openwisp-cotroller (reboot command is working). then follow the steps. in the end you can see that i can deactivate the device. am i missing something ? 

https://github.com/user-attachments/assets/54c3e2f0-2d75-4176-bb82-19082d3f8399

pandafy, 2024-12-16:
You need to keep the VM powered off. The problem arises when the devices are no longer used (i.e. disconnected from the network). When you reboot the VM, OpenWrt downloads the latest configuration from OpenWISP and the configuration status is flagged as deactivated. 

I would recommend you to take up another issue as a solution for this issue would require some brainstorming.

akhilsharmaa, 2024-12-18:
> I would recommend you to take up another issue (look for "good first issue" label). A solution for this issue would likely require some brainstorming before implementing.

Yes, i'm leaving this issue. ; (

dee077, 2025-01-01:
Hey @pandafy,  

I've replicated the bug and enabled the delete button to show when the `config status` is set to **deactivating**. However, I’m unsure about the best way to add the following warning text, which should appear after clicking the delete button:  

> **Warning:**
> "The device is still in the deactivating state, meaning its configuration is still present on the device.
> If you wish to remove the configuration from the device, please wait until [...]

Should I:  
1. Create a new template extending the existing one, `delete_selected_confirmation.html`?  
2. Is there a better approach to integrate this warning?  

Additionally, some guidance on how to edit this warning message would be really helpful.

#### Current Flow:
https://github.com/user-attachments/assets/5b38cb40-23ab-498e-92f8-7ed26136cec5

pandafy, 2025-01-02:
@dee077 can you create a draft PR so we can help you with the changes?

 I think we can update this template [openwisp_controller/config/templates/admin/config/device/delete_selected_confirmation.html](https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/config/templates/admin/config/device/delete_selected_confirmation.html) by adding a conditional (if/else) to show the warning if the device's config status is `deactivating`.

pandafy, 2025-01-02:
Upon further review, this issue looks complicated for new contributors. 

@dee077 I suggest you look for other issues in [OpenWISP Contributor's Board](https://github.com/orgs/openwisp/projects/42).

commit 608cbbfd (2025-01-06): [fix] Allowed deleting device with "deactivating" config status #949

Fixes #949
PR #962 (2025-01-29): [fix] Allowed deleting device with "deactivating" config status #949

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [x] I have updated the documentation.

## Reference to Existing Issue

Fixes #949

## Blockers

- [x] Add more selenium tests 
- [x] Handle deleting multiple deactivating device

https://github.com/user-attachments/assets/ [... cut]
commit 2e5e0dc6 (2025-01-29): [fix] Allowed deleting device with "deactivating" config status #949

Fixes #949

---------
commit caf54204 (2025-01-29): [fix] Allowed deleting device with "deactivating" config status #949

Fixes #949

---------

(cherry picked from commit 2e5e0dc688fa7b9cad0cdd58619870fc88380574)
closed 2025-01-29 by nemesifier