openwisp/openwisp-controller #973: [bug] VPN templates conflict

nemesifier, 2025-01-31:
**Describe the bug**
Assigning a variation of a OpenVPN client template while unassigning another one causes the resulting configuration to become bugged: the configuration variables of the certificates are not resolved.

**Steps To Reproduce**
Steps to reproduce the behavior:
1. Create a VPN-client template using OpenVPN
2. Create another VPN-client template using the same server, but with slightly different configurations
3. Assign the template created in step 1, save
4. Now unassign the template created in step 1 while at the same time assign the template created in step 2, save

**Expected behavior**
The resulting configuration should work normally.

**Screenshots**

[image]

**System Informatioon:**
 - OS: Debian bookworm/sid
 - Python Version: 3.10.12
 - Django Version: 4.2

shwetd19, 2025-02-01:
@nemesifier can I work on this ? would love to move forward with it!

nemesifier, 2025-02-01:
@shwetd19 sure, you can try! Can you replicate the issue?

pandafy, 2025-02-05:
@shwetd19 are you still working on this?

pandafy, 2025-02-05:
This bug occurs due to the following code:

https://github.com/openwisp/openwisp-controller/blob/129a42f957166fe6652e5181e9bf03eadc2c3984/openwisp_controller/config/base/config.py#L279-L298

The m2m field sends the following signals after the operation in **Step 4**:

1. **post_add**: Triggered when adding `template 2`. This creates a new VpnClient object (referred to as VpnClient2) for this configuration, but it also deletes the existing VpnClient object created by `template 1`. Refer to this line:

https://github.com/openwisp/openwisp-controller/blob/129a42f957166fe6652e5181e9bf03eadc2c3984/openwisp_controller/config/base/config.py#L279-L281

2. **post_remove**: Triggered when removing `template 1`. This deletes all VpnClients associated with the VPN related to `template 1`.

https://github.com/openwisp/openwisp-controller/blob/129a42f957166fe6652e5181e9bf03eadc2c3984/openwisp_controller/config/base/config.py#L296-L298

As a result, the configuration ends up with no VpnClient object, causing the VPN configuration variables to remain unresolved.

shwetd19, 2025-02-05:
Hey @nemesifier @pandafy I've made a PR for this can you pls have a look, also I had a question as, Currently the fix focuses on preserving VPN clients when their VPN is still in use by other templates. Do we need additional validation to ensure this doesn't create any inconsistencies with other VPN-related features in OpenWISP?

PR #977 (2025-02-21): [fix] Fix VPN-client template switch bug with same VPN server
Resolves a bug where VPN configuration variables become unresolved when switching between VPN templates using the same VPN server

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- N/A I have updated the documentation.

## Reference to Existing Issue

Closes #973.

## Description [... cut]
commit aba8bebe (2025-02-21): [fix] Fix VPN-client template switch bug with same VPN server #973

Closes #973.

---------
commit 7815ed36 (2025-02-21): [fix] Fix VPN-client template switch bug with same VPN server #973

Closes #973.

---------

(cherry picked from commit aba8bebe24d50572bd0125bf95171bf95bd516d9)