openwisp/openwisp-controller #982: [bug] Change ZeroTier network name from `ow_zt` to `global`

pandafy, 2025-02-28:
### Describe the Bug

This change is necessary because the configuration format has changed on OpenWrt since ZeroTier 1.14.0. Previously, it supported running multiple services of the ZeroTier controller, but now only one service is allowed called `global`. 

We need to update the `Vpn.auto_client` method to use `global` instead of `ow_zt`. 

https://github.com/openwisp/openwisp-controller/blob/aba8bebe24d50572bd0125bf95171bf95bd516d9/openwisp_controller/config/base/vpn.py#L636-L642

We also need to create a database migration which updates the existing templates for ZeroTier VPN backend. It should change the network name in config from `ow_zt` to `global`. 

E.g. if the `template.config` contains the following NetJSON:

```json
{
    "zerotier": [
        {
            "disabled": false,
            "name": "ow_zt",
            "networks": [
                {
                    "id": "<network-id-redacted>",
                    "ifname": "owztbe1234"
                }
            ],
            "secret": "{{secret}}",
            "config_path": "/etc/openwisp/zerotier"
        }
    ]
}
```

Then, after running the migration, the `template.config` should become:

```json
{
    "zerotier": [
        {
            "disabled": false,
            "name": "global", // this is changed from `ow_zt` to `global`
            "networks": [
                {
                    "id": "<network-id-redacted>",
                    "ifname": "owztbe1234"
                }
            ],
            "secret": "{{secret}}",
            "config_path": "/etc/openwisp/zerotier"
        }
    ]
}
```

### System Information
- **OS**: Ubuntu 20.04 LTS
- **Django Version**: Django 4.2

Rehan989, 2025-03-10:
Hi, is this still open? @pandafy

pandafy, 2025-03-10:
@Rehan989 there's already an open PR for this issue.

PR #986 (2025-03-20): [fix] Update zerotier network name from ow_zt to global
## Checklist

- [X] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [X] I have manually tested the changes proposed in this pull request.
- [ ] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [ ] I have updated the documentation.

## Reference to Existing Issue

Fixes #982  .

## Description of Changes

Changed ZeroTier network name from ow_zt to global by modifying migration file and also by updating the Vpn.auto_client met [... cut]
commit 4881326a (2025-03-20): [fix] Updated zerotier network name from ow_zt to global #982

Closes #982
closed 2025-03-20 by nemesifier