openwisp/openwisp-controller #1000: [bug] Paramiko exceptions for SSH network issues shouldn't be logged

nemesifier, 2025-04-02:
When there's network issues processing commands or SSH connections, a number of errors are logged, eg:

[image]

I think some code in OpenWISP Controller influences this too, eg:
https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/connection/connectors/ssh.py#L195-L197

It would be great if we can avoid this.

praptisharma28, 2025-04-07:
working on this one !

pandafy, 2025-04-27:
We can replicate this scenario by rejecting SSH packets using iptables. I used the following command on OpenWrt 24.10

```bash
iptables -I INPUT -i wg0 -p tcp --dport 22 -m conntrack --ctstate ESTABLISHED -j REJECT --reject-with tcp-r
eset
```

> [!NOTE]  
> The above rule only rejects SSH packets on the `wg0` interface. I purposefully did this to avoid loosing SSH connection to the device. 

These errors are not generated by the OpenWISP Controller itself, but rather by Paramiko (see the [Paramiko source code](https://github.com/paramiko/paramiko/blob/ed8b09751ff20340332d4b1bb2b10e32aedc57ff/paramiko/transport.py#L2281-L2306) for the relevant portion). Due to the way Paramiko formats its traceback strings, each line in the stack trace is treated as a separate event in Sentry, which is why they appear as individual events (see https://github.com/paramiko/paramiko/issues/2366).

Given that we have robust error handling in place for all SSH connections, I believe these errors are not critical and can be safely ignored. To address this, I've opened https://github.com/openwisp/openwisp-sentry-utils/pull/8 to configure Sentry to ignore these specific messages.

closed 2025-05-02 by nemesifier