openwisp/openwisp-controller #1016: [bug] Command API allow creating command for devices without DeviceConnection

pandafy, 2025-05-01:
**Describe the bug**
The command API allows creating command for devices that do not have a DeviceConnection object (without assigned credentials). 

I conflicts with the UI implementation which only allow sending commands if the device has a DeviceConnection object. 

**Steps To Reproduce**
Steps to reproduce the behavior:
1. Create a device without assigned credentials
2. Go to `http://127.0.0.1:8000/api/v1/controller/device/<device-id>/command/`
3. Create a `reboot` command with the following payload 
```
{
    "input": null,
    "type": "reboot"
}
```

**Expected behavior**
The REST API should return 400 response complaining that the device does not have any credentials assigned.

DragnEmperor, 2025-05-05:
Hi @pandafy, I’ve looked into this and implemented a solution. Would you like me to open a PR for it?

pandafy, 2025-05-05:
> Hi [@pandafy](https://github.com/pandafy), I’ve looked into this and implemented a solution. Would you like me to open a PR for it?

Sure, go ahead!

commit 4836913a (2025-05-23): [fix] Prevent command creation for devices without a DeviceConnection #1016

Verify that a device has associated DeviceConnection
credentials before allowing to create new Command objects.

Additionally, enforce validation of connection and command types
to ensure only permitted values are diasplayed and accepted
through the REST API.

Closes #1016