openwisp/openwisp-controller #1025: [feature] Set Up WHOIS Data Retrieval and Storage for Devices

DragnEmperor, 2025-05-11:
This involves:
- Setting up geoip2 database
- Setup WHOIS model to store required information : organization name, country, ASN, address, CIDR
- Implement Celery Task to automatically fetch the required details on basis of last public ip address reported by device
- Addition of these details on device admin alongside last_ip

DragnEmperor, 2025-05-12:
There are two ways to fetch the data using geoip2 : 
- Through web service (requires internet for each fetch)
- Through manually downloading the database (requires scripts for updating)

Let me know your thoughts!

devkapilbansal, 2025-05-14:
> * Through manually downloading the database (requires [scripts](https://dev.maxmind.com/geoip/updating-databases/) for updating)
I believe [Django also advices](https://docs.djangoproject.com/en/5.1/ref/contrib/gis/geoip2/#module-django.contrib.gis.geoip2) to download the data. How frequently we may need to update the scripts?

DragnEmperor, 2025-05-15:
> * Through manually downloading the database (requires [scripts](https://dev.maxmind.com/geoip/updating-databases/) for updating)
>   I believe [Django also [...]

[Maxmind](https://www.maxmind.com/en/geoip-databases) updates the databases twice a week, every friday and tuesday.
[DB-IP](https://db-ip.com/db/) databases are updated monthly

Also DB-IP databases do not require an account creation, while Maxmind databases does.

codesankalp, 2025-05-20:
> There are two ways to fetch the data using geoip2 :
>
> * Through [web service](https://geoip2.readthedocs.io/en/latest/#sync-web-service-example) (requires internet for each fetch)
> * Through manually [...]

I’m also in favor of proceeding by downloading the database, with the added flexibility to configure and switch the database source when needed.

DragnEmperor, 2025-05-20:
## **More information regarding services provided by Maxmind:**
### Webservices

- GeoLite2 Web services are free but are Limited to 1000 requests per day
- GeoIP2 Web services are paid and starts with 20$ (10,000 queries amounting to 0.002$ per query)
- Both these services do not require any additional licensing

More info can be found here: 
https://www.maxmind.com/en/geoip-api-web-services#buy-now

### Databases

- GeoLite2 Databases are free to download but requires [commercial licensing](https://support.maxmind.com/hc/en-us/articles/4408928143643-Commercial-Redistribution-License-for-GeoLite2). The total size can vary between 50 - 70 mb.
- GeoIP2 Databases are paid with starting price of 134$ (we require city database as the fields we require are present in city db only)

More References:
https://dev.maxmind.com/geoip/docs/databases/city-and-country/#database-sizes
https://www.maxmind.com/en/site-license-overview
https://www.maxmind.com/en/geoip-databases

### **More information regarding services provided by DB-IP:**
### Databases
Only databases are free which are updated monthly with total size of 100 - 130mb

DragnEmperor, 2025-05-21:
> Hi [@DragnEmperor](https://github.com/DragnEmperor) As discussed in the call today, although databases is a preferred solution, for now, we will proceed with the APIs as they will be less time taking to implement.
> 
> You can create a separate task for database integration which we can look into once our deliverables are done

> > Hi [@DragnEmperor](https://github.com/DragnEmperor) As discussed in the call today, although databases is a preferred solution, for now, we will proceed with the APIs as they will be less time taking to implement.
> > You can create a separate task for database integration which we can look into once our deliverables are done
> 
> Thanks [@DragnEmperor](https://github.com/DragnEmperor) for the info, I agree with Kapil to create a separate issue for this and copy that valuable information you have shared there to the issue description.

Have created an issue for this : https://github.com/openwisp/openwisp-controller/issues/1052
As discussed, keeping it separate from the main GSoC project. We can pick this later as per requirement and demand.