openwisp/openwisp-controller #1033: [feature] Implement Celery task for WHOIS fetch based on device last IP

DragnEmperor, 2025-05-14:
After closing of #1032, we need to implement celery task which will trigger whenever `last_ip` changes and `is public`. 
The task will be responsible for fetching the required details as per the approach finalized for #1032.
The details will then update/create the WHOIS details of that device

DragnEmperor, 2025-05-21:
Following are the items which I think we can discuss before implementing this to have more clarity:

- Should we add a retry mechanism if task fails?
- Are we providing a way for user to know if the task has succeeded or not? Or if it fails then let the user know of the user or something that might help in debugging?

devkapilbansal, 2025-05-26:
> Should we add a retry mechanism if task fails?

Definitely, this retry mechanism should be kept configurable

> Are we providing a way for user to know if the task has succeeded or not? Or if it fails then let the user know of the reason or something that might help in debugging?

Probably, we can add a graph for this.

DragnEmperor, 2025-05-27:
> > > > Are we providing a way for user to know if the task has succeeded or not? Or if it fails then let the user know of the reason or something that might help in debugging?
> > > 
> > > 
> > > Probably, we can add a graph for this.
> > 
> > 
> > I was thinking of sending a notification to user of the failed task, and as for debugging exposing and endpoint that fetches task status. Something like :
> > class TaskStatusView(APIView):
> > def get(self, request, taskID):
> > result = AsyncResult(str(taskID))
> > payload = {"taskID": taskID, "status": result.status}
> > if result.successful():
> > payload["result"] = result.result
> > elif result.failed() and hasattr(result, "result"):
> > payload["result"] = str(result.result)
> > return Response(payload)
> > As for the graph, do you mean visualizing a chart or something like that for showing success vs failed tasks? Sorry if I got that wrong.
> 
> I am not following this. We decided to send a notification to the user for creation of a fuzzy location. Maybe, we can send a web notification if the task fails?
> 
> For debugging, the users (OpenWISP managers) should be able to lookup the logs on the server.

This is a sample code for returning the status of task and its result if task failed or succeeds.
Yes, we don't need this if server logs can be accessed.

I think I can proceed with an implementation of the failure mechanisms. Thanks @devkapilbansal @pandafy

commit fc6b0d2c (2025-05-29): [feature] Base setup for WHOIS model #1032 #1033

Added WHOIS model with the required fields.
Implemented a new celery task to fetch WHOIS
details using the `geoip2` web service.
In order to trigger the task whenever `last_ip`
changes, using existing logic of `_changed_fields`
to track the changes.

Closes #1032
Closes #1033
commit 9de53676 (2025-06-16): [feature] Base setup for WHOIS model #1032 #1033

Added WHOIS model with the required fields.
Implemented a new celery task to fetch WHOIS
details using the `geoip2` web service.
In order to trigger the task whenever `last_ip`
changes, using existing logic of `_changed_fields`
to track the changes.

Closes #1032
Closes #1033
commit 2d18e68c (2025-06-23): [feature] Base setup for WHOIS model #1032 #1033

Added WHOIS model with the required fields.
Implemented a new celery task to fetch WHOIS
details using the `geoip2` web service.
In order to trigger the task whenever `last_ip`
changes, using existing logic of `_changed_fields`
to track the changes.

Closes #1032
Closes #1033
PR #1054 (2025-06-30): [feature] Base setup for WHOIS model #1032 #1033 #1037 #1045
## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [x] I have updated the documentation.

## Reference to Existing Issue

Closes #1032
Closes #1033
Closes #1037
Closes #1045 

## Description of Changes

Added a new model to store WHOIS details for a device with the required fields. [... cut]