openwisp/openwisp-controller #1035: [enhancement] Trigger notification on update/create fuzzy location and show alert

DragnEmperor, 2025-05-14:
Whenever a fuzzy location is created/updated, a notification should be triggered to user as per the preferred mode (Web or Email or Both) with proper information.
The fuzzy location page should also display an alert or use some UI indicator informing user of the fuzzy nature of location and encourage them to manually update/refine it.

pandafy, 2025-05-20:
**Thinking out loud:**

Writing down some questions to brainstorm about the UX

- How often will this notification be sent?
- Do we need to send an email notification for this operation? 
- Can we use the `generic_message` notification type for this notification which only sends web notification?

DragnEmperor, 2025-05-21:
> **Thinking out loud:**
> 
> Writing down some questions to brainstorm about the UX
> 
> * How often will this notification be sent?
> * Do we need to send an email notification for this operation?
> * Can we use the `generic_message` notification type for this notification which only sends web notification?

For first one, I think we can have something like threshold period, based on created/updated field of location to check if it is again modified within the threshold period then do not trigger.

For second one, yes sending email will seem like spam. But if it is required then we can send it for creation and not updation

For third one, yes we can use this for web only notifications.

DragnEmperor, 2025-06-05:
> > Instead of showing an alert, we can explore something along the side of django messages framework, to display a warning on the location admin page indicating the fuzzy nature.
> 
> What if the network administrator is not online when the fuzzy location is created? What shortcoming do you see with using notifications?

Hi @pandafy ,
We are sending notifications whenever a fuzzy location is created/updated. The above is for when a user opens admin of a fuzzy location, in that case we must show some kind of warning/information indicating that the location is fuzzy and they should optimize it manually.

commit a9c5fbc0 (2025-08-20): [feature] Estimated location: added notification and admin warning #1035

Closes #1035
commit 0b389a68 (2025-08-29): [feature] Estimated location: added notification and admin warning #1035

Closes #1035
commit 8aca0dcc (2025-09-09): [feature] Estimated location: added notification and admin warning #1035

Closes #1035
commit dbb78859 (2025-09-21): [feature] Allow estimating geographic location based on WHOIS data #1034

- Estimated geographic location logic, closes #1034.
- Added notification and admin warning, closes #1035.
- Set location estimated flag to ``False`` on manual update,
  closes #1036.
- Added admin and API filters for estimated locations,
  closes #1028.

---------