openwisp/openwisp-controller #1046: [tests] Test cases ensuring correct working of Fuzzy Location feature

DragnEmperor, 2025-05-16:
Test cases ensuring correct creation and handling of fuzzy locations based on WHOIS lookup. Test coverage should include:
- Creation of new fuzzy location and updating the same whenever last_ip of device changes.
- Ensuring `is_fuzzy` is set True/False when auto-generated via task or when manually edited via admin or API respectively.
- Notification is triggered after fuzzy location creation/updated.
- Alert is shown when viewing fuzzy location in the admin interface.
- Behavior when organization.is_fuzzy_enabled=False (no fuzzy location should be created).

**Dependencies**

- [ ] https://github.com/openwisp/openwisp-controller/issues/1034
- [ ] https://github.com/openwisp/openwisp-controller/issues/1035
- [ ] https://github.com/openwisp/openwisp-controller/issues/1036
- [ ] https://github.com/openwisp/openwisp-controller/issues/1037

nios-x, 2025-12-26:
Hello,
My name is Soumya and I am a student and an open-source enthusiast. I would love to participate in gsoc and work on #1046 – Test cases to ensure that Fuzzy Location is working correctly. Could you assign it to me.

DragnEmperor, 2025-12-26:
@nios-x This issue is part of GSoC 2025 program and is completed. Once the main PR is merged, all the related issues (including this one will be marked as done). You can check out other issues which do not have the `gsoc` label attached to it.

nios-x, 2025-12-26:
Means I cannot contribute to this repo for GSoC  2026 anymore?