openwisp/openwisp-controller #1071: [bug] Device group filter in admin shows "shared systemwide" instead of "no group"

nemesifier, 2025-06-20:
**Describe the bug**

[image]

When trying to filter devices which have no group assigned, after selcting the "-" option in the group filter, the UI shows "Shared systemwide".

**Steps To Reproduce**

1. Go to device list
2. Click on "By Group" filter
3. Select "-"
4. The label "Shared systemwide" appears

**Expected behavior**
It would be better to either keep showing "-" or something more user friendly like N/A.

**Screenshots**

https://github.com/user-attachments/assets/3cccb561-d22c-4f2f-bb15-ff0eefd0b190

panagiotisevaggelou, 2025-09-25:
Hello @nemesifier, I’m a new contributor and I’ve opened a PR that addresses this issue. 
Would appreciate any feedback or suggestions. Thanks!

nemesifier, 2025-09-27:
> Hello [@nemesifier](https://github.com/nemesifier), I’m a new contributor and I’ve opened a PR that addresses this issue. Would appreciate any feedback or suggestions. Thanks!

Will look into it asap, thanks!

panagiotisevaggelou, 2025-09-30:
Thank you! I’ll keep an eye out for more issues to contribute to.

PR #1129 (2025-09-30): [fix] Fixed device group filter label in device list #1071
## Checklist

- [X] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [X] I have manually tested the changes proposed in this pull request.
- [ ] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [ ] I have updated the documentation.

## Reference to Existing Issue

Closes #1071

## Description of Changes

This PR fixes the label shown in the device group filter on the device list view.

Previously, when selecting the “no gro [... cut]
commit ac6fd838 (2025-09-30): [fix] Fixed device group filter label in device list #1071

Replaced "Shared systemwide (no organization)" with "-" in the
device group filter dropdown of the device list view, while keeping
the original label unchanged elsewhere.

Fixes #1071