openwisp/openwisp-controller #1100: [bug]

Freak4915, 2025-07-27:
VLAN Filtering configuration appears at the template level only and not in the final device configuration and therefore never applied.

**Steps To Reproduce**
Steps to reproduce the behavior:
Go to Configurations -> Templates -> Add template
Add configuration item Interfaces
Add an interface, and fill in details needed.
Add VLAN Filtering
Add VLAN 1
Add ports lan1
Tagging Egress tagged
Tick Primary VID
Add ports lan2
Tagging Egress tagged
Tick Primary VID
Add ports lan3
Tagging Egress tagged
Tick Primary VID

Check Preview configuration and confirm:

config bridge-vlan 'vlan_br_lan_1'
	option device 'br-lan'
	list ports 'lan1:t*'
	list ports 'lan2:t*'
	list ports 'lan3:t*'
	option vlan '1'

**Expected behavior**
Go to Devices
Select Device and apply the created template.
Click on Preview configuration

All details about the just add template appears but the VLAN Filtering is missing.

**Screenshots**

[image]

**System Informatioon:**
 - OS: Ubuntu lsb_release -a
 - Python Version: Python 3.12.3
 - Django Version: [e.g. Django 4.2.5]
 - Browser and Browser Version (if applicable): Edge

192.168.0.165

Freak4915, 2025-07-29:
netjsonconfig 1.2a0 appears to be currently installed, but I don't have this option in the backend dropdown.

Freak4915, 2025-07-29:
`package network

config device 'device_br_lan'
	option ageing_time '300'
	option forward_delay '4'
	option hash_max '512'
	option hello_time '2'
	option igmp_snooping '0'
	option last_member_interval '100'
	option max_age '20'
	option mtu '1500'
	option multicast_querier '0'
	option name 'br-lan'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	option priority '32767'
	option query_interval '12500'
	option query_response_interval '1000'
	option robustness '2'
	option stp '0'
	option type 'bridge'
	option vlan_filtering '1'

config bridge-vlan 'vlan_br_lan_1'
	option device 'br-lan'
	list ports 'lan1:t*'
	list ports 'lan2:t*'
	list ports 'lan3:t*'
	option vlan '1'

config bridge-vlan 'vlan_br_lan_40'
	option device 'br-lan'
	list ports 'lan1:t'
	list ports 'lan2:t'
	list ports 'lan3:t'
	option vlan '40'

config interface 'vlan_br_lan_1'
	option device 'br-lan.1'
	option proto 'none'

config interface 'vlan_br_lan_40'
	option device 'br-lan.40'
	option proto 'none'

config interface 'br_lan'
	option auto '1'
	option device 'br-lan'
	option enabled '1'
	option proto 'none'

config device 'device_vlan1'
	option ageing_time '300'
	option forward_delay '4'
	option hash_max '512'
	option hello_time '2'
	option igmp_snooping '0'
	option last_member_interval '100'
	option max_age '20'
	option mtu '1500'
	option multicast_querier '0'
	option name 'br-vlan1'
	list ports 'br-lan.1'
	option priority '32767'
	option query_interval '12500'
	option query_response_interval '1000'
	option robustness '2'
	option stp '0'
	option type 'bridge'

config interface 'vlan1'
	option auto '1'
	option device 'br-vlan1'
	option enabled '1'
	option proto 'none'

config device 'device_vlan40'
	option ageing_time '300'
	option forward_delay '4'
	option hash_max '512'
	option hello_time '2'
	option igmp_snooping '0'
	option last_member_interval '100'
	option max_age '20'
	option mtu '1500'
	option multicast_querier '0'
	option name 'br-vlan40'
	list ports 'br-lan.40'
	option priority '32767'
	option query_interval '12500'
	option query_response_interval '1000'
	option robustness '2'
	option stp '0'
	option type 'bridge'

config interface 'vlan40'
	option auto '1'
	option device 'br-vlan40'
	option enabled '1'
	option proto 'none'
`

Freak4915, 2025-07-29:
This is just one of 11 temples applied. If you want the full device template, please let me know, but essentially it includes the above but missing:

config bridge-vlan 'vlan_br_lan_1'
option device 'br-lan'
list ports 'lan1:t*'
list ports 'lan2:t*'
list ports 'lan3:t*'
option vlan '1'

config bridge-vlan 'vlan_br_lan_40'
option device 'br-lan'
list ports 'lan1:t'
list ports 'lan2:t'
list ports 'lan3:t'
option vlan '40'

nemesifier, 2025-09-30:
This should have been fixed by https://github.com/openwisp/netjsonconfig/pull/363. If after upgrading to the latest dev versions both openwisp-controller and netjsonconfig the issue still appears, please reopen the issue and provide the JSON config (from advanced mode) which allows to replicate the issue.

closed 2025-09-30 by nemesifier